Matthew1471/Enphase-API #1: Document which software version the structures apply to

dlmcpaul, 2023-06-27:
Good work documenting all the structures.  It looks like you are working with V7

Just be aware that older versions do not have all the fields and/or API's.  Might want to look at making that clear what versions they apply to.

Matthew1471, 2023-06-27:
Thank you, yes that's correct as Authentication Tokens are only V7 specific too...

Unfortunately I do not have any other hardware to compare my V7 against (although as APIs get further added or removed I will be able to update the documentation) - so I can't go back in time and highlight what endpoints used to exist or their response format.

If you have any examples I'd welcome any additions to the documentation. The endpoint documentation is currently automatically generated by a script in the "examples" folder, `gateway_generate_docs.py` so any "Version information" would need to be added to the generating script rather than manually editing each of the .adoc files (they'll get over-written the next time the script is ran otherwise).

Welcome any further feature requests/ideas also :)

dlmcpaul, 2023-06-27:
I wrote my own monitor software for Enphase (See https://github.com/dlmcpaul/EnphaseCollector)

I have a set of test cases going back to 4.2.27 (See https://github.com/dlmcpaul/EnphaseCollector/tree/master/src/test/resources/stubs for the responses)

Not perfect since I did not own all the variations so some of the responses are not 100%.  When someone has trouble they send me what they can that matches the API calls I make.  I don't use the streaming or installer API's though.

The hardest piece everyone is trying to understand is the battery stats.  It looks like there is a lot of variation there.

Also Authentication Tokens look to vary depending on the sub version in the 7 line.  I think they have removed the need to call the /auth/check_jwt endpoint but it has not been disabled.

Matthew1471, 2023-06-27:
>I think they have removed the need to call the /auth/check_jwt endpoint but it has not been disabled.

They added helper functionality that some methods allow you to call them with a bearer token and internally that will automatically call check_jwt.. they haven't added it to all.

If you're making multiple calls to the Gateway it would be more performant to manually call check_jwt and create a session than getting it to keep checking JWTs.

> I don't use the streaming or installer API's though.

I am aware that on some models the streaming URL is restricted to installer only tokens. There was a community forum post about this and Enphase have promised to return the streaming URL to non-installers in a future release. I'm running a version which is not fussy about the type of token for the streaming URL. The advantage to the streaming URL is that the gateway does not have to repeatedly parse requests.. I definitely recommend using the streaming URL if you can.

I've seen https://github.com/dlmcpaul/EnphaseCollector/issues/55 and will comment with some points on that there.