Matthew1471/Enphase-API #4: Add Phasemode to ivp/meters

catsmanac, 2023-09-03:
[ivp/meters](https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/IVP/Meters/EID.adoc) field phaseMode is currently described as 'The number of phases the meter can monitor' which is for phaseCount. 

I know of at least:

- three
- split

values showing in phaseMode.

Matthew1471, 2023-09-03:
Thanks, since making this I've come across the source code for the endpoints and am in the process of adding the details to this project (including auth levels).

I've checked and the value can be `single`, `split` or `three`.

I have the Enphase Gateway IQ Metered which is capable of monitoring triple phase (and mine reports `three`) but I am in a single phase setup, which is why I put the comment that it's what it *can* monitor (rather than what it is set to monitor).

It looks like it's set in the firmware as an environment variable /etc/enphase.conf and is more a property of the device than something user configurable:

```
export EMU_PART_NUMBER=800-00654-r08
export EMU_SERIAL_NUMBER=12224xxxxxxx
export EMU_REGION=EU
export EMU_METERING_MODE=three
export EMU_FREQ_PLAN=103_110_116kHz
```

and if I were to guess I would say that "split" is where you have it monitoring multiple lines but those lines don't form to make a triple phase setup or they aren't to include the production.

`/ivp/meters/*` is actually serviced internally by the `eim` process rather than the `emu` endpoint that most of the other API is serviced by.

I'll update the docs to include `single`, `split` or `three`.

Matthew1471, 2023-09-03:
Commit 1253ebe53f8a0520236772b8bfc8b3d22b526e7c resolves this issue, please see updated documentation for [/IVP/Meters/EID](https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/IVP/Meters/EID.adoc) and [/IVP/Meters/All](https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/IVP/Meters/All.adoc)

closed 2023-09-03 by Matthew1471

catsmanac, 2023-09-03:
Thanks for the update and explanation @Matthew1471.

>/ivp/meters/* is actually serviced internally by the eim process rather than the emu endpoint that most of the other API is serviced by.

These are reportedly faster too. The /production page sometimes takes multiple seconds and some have reported is not responding an a daily base at 23:00 for 1 minute.

We were mapping fields/values between /production and ivp/meters/readings and see the same numbers with different names as well as some different fields reported by each.

Matthew1471, 2023-09-03:
You're welcome. Yes having looked at the source code for `/production` it makes a LOT of other API calls. I also was surprised to find that `/stream/meter` internally was just calling `/ivp/meters/reports` on a 0.99 timer.

Also the IQ Gateway has quite a few web servers which various calls jump through, so when it makes an internal call it ends up amplifying the load rather than if you call directly and take the load hit once.

catsmanac, 2023-09-03:
>I also was surprised to find that /stream/meter internally was just calling /ivp/meters/reports on a 0.99 timer.

Ah interesting. so both ivp/meters/readings and ivp/meters/reports are served by that same process? These 2 pages use different names for what seems to be the same data.

Historically many data collectors used /production page and that provides the current and last 7 days energy numbers. The meters page offer alternatives for lifetime and current power, but not for these daily and last seven days energy numbers, so can't replace the production page without breaking these. Do yo know alternatives for these?