Matthew1471/Enphase-API #9: Looking for an battery energy endpoint

Hoffmann77, 2023-11-02:
Hey Matthew1471,
I really appreciate your effort maintaining a documentation for the API.

Are you by any chance aware of an endpoint that provides data for the total energy charged/discharged by Encharge batteries? I'm only aware of the endpoints providing the instantaneous power of the batteries.
Would be really nice to add that data to the home-assistant Enphase integrations.

Best regards

TobiasNeuschulz, 2025-04-22:
Please check:

https://<IP-address>/ivp/ensemble/power
https://envoy.local/ivp/ensemble/power

These endpoint shows the state of each battery: current State of Charge and charge/discharge rates.

I don't see these endpoints in this API documentation.

NathanSweet, 2025-07-23:
/ivp/ensemble/power:
```
{
	"devices:": [
		{
			"serial_num": "xxx",
			"real_power_mw": 0,
			"apparent_power_mva": 0,
			"soc": 100
		},
...
```

This is instantaneous though.