Matthew1471/Enphase-API #10: Documentation for fresh solar install on how to begin leveraging this repo

Guyverix, 2023-11-17:
I just got my solar activated yesterday, and am looking to start monitoring locally with my servers.  I cannot seem to find a good howto for how exactly to  get the auth credentials so I can talk to the system..

I am guessing this part from the quick start guide:
```
3. Auth URL
To start making API requests, the application must be authorized by the system owner. The Auth URL must be sent to the system owner via email or must be embedded in the HO App created by the developer. For the above app, Auth URL is https://api.enphaseenergy.com/oauth/authorize?response_type=code&client_id=f2a479f3c6067f0d9517cadae7f00b47
```
But who do I reach out to at Enphase for this?  Do we have to imply we are creating an app for them to give us credentials?

I am trying to be honest with Enphase, and am really hoping that there is somewhere I can simply request auth keys so I can query my local install and drop data into Grafana and Graphite/InfluxDb on my own.

cgarwood, 2023-11-18:
If everything is local, this document should give you the info you need to get started: https://enphase.com/download/accessing-iq-gateway-local-apis-or-local-ui-token-based-authentication

Guyverix, 2023-11-18:
Thank you very much!

closed 2023-11-18 by Guyverix

Matthew1471, 2023-12-03:
So there's a few things confused here.

Official Cloud Enphase API - have to get a key from them on their developer site.

Local IQ Gateway API - no need to do anything special other than follow my quick start https://github.com/Matthew1471/Enphase-API/tree/main/Documentation/Wrappers/Python as this library will get a token from Enphase that if you are a System Owner will be valid for 1 year.

The document posted above shows how you can access it via curl etc.. this Python project provides all the functionality to offer via Python in your code and a few helpful examples (to write the data into AMQP/MQTT or a SQL database or shown on a Raspberry Pi display)

Hope this helps!