Matthew1471/Enphase-API #12: Getting inverter-level power, current, voltage and temperature from the local API?

PertuyF, 2023-12-13:
Hi there,
It looks like enlighten is able to retrieve quite detailed information at the inverter level, based on [this article](https://support.enphase.com/s/article/How-do-I-get-the-microinverter-level-graph-showing-voltage-current-frequency-and-temperature).
I wasn't able to find information about this in  the doc from this repo. Would anyone have an idea how to get this data?

Best,
Fabien

Matthew1471, 2023-12-13:
Try https://github.com/Matthew1471/Enphase-API/blob/a9a2c14f7fed228f43bbfde6c44196f2183ed17d/Documentation/IQ%20Gateway%20API/IVP/PEB/DevStatus.adoc

and https://github.com/Matthew1471/Enphase-API/blob/a9a2c14f7fed228f43bbfde6c44196f2183ed17d/Documentation/IQ%20Gateway%20API/V1/Production/Inverters.adoc#L42

PertuyF, 2023-12-13:
`DeviceStatus` is exactly what I have been looking for, and I missed the details in the doc.
Thanks a ton for your help, and for your work putting this project together!

Matthew1471, 2023-12-13:
You're welcome, closing the issue 😄

closed 2023-12-13 by Matthew1471

PertuyF, 2023-12-16:
@Matthew1471 I'm having troubles with `DeviceStatus`. I keep getting `401: Unauthorized` despite using a valid token and reusing the session cookies from `/auth/check_jwt`.
I may be doing it wrong, any tips welcome :)

Matthew1471, 2023-12-16:
> @Matthew1471 I'm having troubles with `DeviceStatus`. I keep getting `401: Unauthorized` despite using a valid token and reusing the session cookies from `/auth/check_jwt`.
> 
> I may be doing it wrong, any tips welcome :) 

Requires an Installer level token (technically anything equal or above "Support" which is Level 4) rather than an Owner one.

Enphase recently started locking down the API dependent on what "role" your JWT token has.

Get a token from an email address that Enphase thinks is an installer and that has access to your Enphase System (the owner has granted access). Installer tokens expire every 12 hours, my code can help with renewing those tokens.

PertuyF, 2023-12-18:
Got it to work, thanks a lot for your guidance :)