Matthew1471/Enphase-API #24: JWT without parsing

NathanSweet, 2024-06-22:
This project reports Enphase now provides an endpoint to get the JWT without janky parsing:
https://github.com/evilzenscientist/envoyAPI?tab=readme-ov-file#updated-20-june-2022
Script showing it is here:
https://github.com/evilzenscientist/envoyAPI/blob/main/entrez-local.ps1

Matthew1471, 2024-07-07:
Thanks for the link, we implement all these functions already.

There's already `get_token_from_enlighten_session_id()` inside `authentication.py`.. however I am not a fan of having a dependency on Enlighten AND the auth server. The parsing is no more brittle than any other URL or querystring Enphase could change.

There is also `get_token_from_oauth()` if you prefer to interact with the auth server without parsing and gateway.py has the sister code `login_oauth_code()`.. this is currently baked into the existing IQ Gateway login flow when you visit the Web GUI so I believe this would be the best way to login without parsing.

NathanSweet, 2024-07-07:
I see, thanks for the explanation. I agree it's nice to only hit the auth server (entrez). I'll give oauth a try.

closed 2024-07-07 by NathanSweet