Matthew1471/Enphase-API #26: stream/meter sever

egomes951, 2024-07-06:
Hello,

I have been using my Enphase solar system for a few years without any major problems. To monitor the system, I am using Node Red. Since my Envoy is running version D7.6.175 (f79c8d), I have to go through the entire process of authentication and obtaining tokens, and every minute I am using the "/ivp" API /meters/readings" to get values.

I'm trying to use the events sent by the stream/meter server, but I always get a 401 unauthorized error message, even though I can see it perfectly in the browser.

For testing purposes I am using cURL.

Can someone help me please ?

Thanks
EG

Matthew1471, 2024-07-07:
Hello,

"[/stream/meter](https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/Stream/Meter.adoc)" has been restricted to installers only in newer firmware.. there's info on that in [this project's documentation](https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/Stream/Meter.adoc)

As for [/ivp/meters/readings/](https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/IVP/Meters/Readings/All.adoc) that should be working fine.. do the examples work perhaps [the MySQL one(https://github.com/Matthew1471/Enphase-API/blob/main/Python/examples/gateway_database_meters.py)?

closed 2024-07-08 by Matthew1471

egomes951, 2024-07-08:
Hi, 

In the browser it works fine, I just can not reproduce it using cURL

[image]

Matthew1471, 2024-07-08:
And you are using the same Enlighten e-mail address and password with the Python examples in this project?

egomes951, 2024-07-09:
(...) and are you using the same email address and password as Enlighten? (...)
-> Yes, using the same credentials

(...) With curl, you will need to at least try to add the JWT token (although you will probably have to send it to the check_jwt endpoint first and then get a sessionId and provide it as a cookie to the stream endpoint - not all Enphase endpoints in all versions of IQ Gateway support a bearer token being provided automatically).. (...)
--> When I call the API https://<envoy>/auth/check_jwt to verify the token, I save the cookie with the sessionId to a file to use later.

with this cookie I able to call the production API without passing the without passing the Authorization Bearer header.

The strange thing is if I copy the sessionID form the browser and put it in the cURL command like this "curl -k https://envoy/stream/meter --cookie "sessionId:<sessionid>" it works, but not with the sessionID that i obtain with the cURL command to validate the token:
curl -k -i -H "Accept: application/json" -H "Authorization: Bearer <token>" "https://envoy/auth/check_jwt" -b cookie -c cookie

egomes951, 2024-07-09:
Doing some more research, the token obtained using the cURL commands is different from the one obtained using the web page.

I've just tested to access the stream/meter SSE using the token obtaind by cURL and I've got the same unauthorized error.

Matthew1471, 2024-07-10:
Don't share the signature but jwt.io will show you the payload to inspect. You should be able to work out the type of tokens being issued via that. I suspect the tokens that work are "installer" and the ones that don't are not.