Matthew1471/Enphase-API #29: Add /ivp/pdm/device_data

krbaker, 2024-11-28:
This endpoint seems mostly undocumented online.  It seems to have the detailed per-micro-inverter data (DC V, DC A, Frequency, Temperature) and is available with a regular customer account only.  I gather there is also more data about the battery systems (though I don't have one to test myself)

Matthew1471, 2024-12-08:
Enphase haven't fixed a security vulnerability I reported so I have full access to all the endpoints.. I'll be updating the documentation to include all the endpoints I am aware of (even the dangerous ones) shortly.

catsmanac, 2024-12-09:
if it helps, here's one for Envoy EU not metered 8.2.4264: [ivp_pdm_device_data.txt](https://github.com/user-attachments/files/18058701/ivp_pdm_device_data.txt)

Note the limit at the bottom of the list.

Bidski, 2025-05-05:
Is there any update to this? I would very much like to be able to monitor the DC details from the microinverters