Matthew1471/Enphase-API #31: Power Export Limit (PEL) API Behavior

jurrid, 2025-03-20:
I was recently in contact with my installer regarding the Power Export Limit (PEL) functionality. They initially informed me that it was not possible to toggle the limiter on or off manually. However, they were able to enable it for me.

For context; I only have a HomeOwnerAccount, not an InstallerAccount, for context.

Due to an error by applying the settings, my installer accidentally sent me a technical screenshot with a API call, which led me to investigate further. After experimenting with the API, I successfully turned off the PEL settings myself using the following API call:

```bash
curl --request POST \
  --url https://x.x.x.x/ivp/ss/dpel \
  --header 'Accept: application/json' \
  --header 'Authorization: Bearer **TOKEN**' \
  --header 'Content-Type: application/json' \
  --data '{
	"dynamic_pel_settings": {
		"enable": false
	}
}'
```

However, when attempting to re-enable the PEL using the following API request, I encountered an error:

```
curl --request POST \
  --url https://x.x.x.x/ivp/ss/dpel \
  --header 'Accept: application/json' \
  --header 'Authorization: Bearer **TOKEN**' \
  --header 'Content-Type: application/json' \
  --data '{
	"dynamic_pel_settings": {
		"enable": true,
		"export_limit": 100
	}
}'
```

Additional Findings
The installer initially said toggling PEL was impossible, but after confirming with them, they acknowledged that I had successfully turned it off. They escalated my case to Dutch Enphase Customer Support, who mentioned that this behavior is an ”unintended flaw” in Envoy v8 but claimed it is “not normal behavior.” to toggle this with an api.

It might be worth further investigation this undocumented API behavior.

fobrs, 2025-07-16:
I use the same commands to toggle PEL, and those work ok.

Specify power after limit_value_W

I use as json:

> {
    "dynamic_pel_settings": {
        "enable": true,
        "export_limit": false,
        "limit_value_W": 3500.0,
        "slew_rate": 900.0,
        "enable_dynamic_limiting": false
    },
    "filename": "site_settings",
    "version": "00.00.01"
}

fobrs, 2025-07-18:
Did you use your token? What is the firmware version of your envoy (you need 8.x.y.z)

UseN1bble, 2025-07-18:
Okay, I made a stupid mistake, and did a GET instead of POST.
Unfortunatly I get the same problem as with stopping production.

I get a 401 Authorization Required message.
Both with my Home as with my Installer Account.

My installer (who is bankrupt) seems to have done something with my installation, that even Enphase can't solve as it seems.

Really frustrating to not be able to limit production while it should work and has worked!

My current version is: D8.3.5167 (5c3a00)

jurrid, 2025-07-18:
In my case, the installer had to activate a certain "feature", and after that, I was able to toggle it. Maybe that feature is still disabled in your setup, but that’s a wild guess.

fobrs, 2025-07-18:
Do you have a Metered envoy with CT's installer?

UseN1bble, 2025-07-19:
Yes, technically there shouldn't be a problem. It's a rights issue.

With my installer account, through the Enlighten portal, I can turn off production and set export limits. Both don't work, but after the update to version 7, I was able to turn off production through the local API. After 4 weeks or so, it didn't work anymore. It's really frustrating and apparently, Enphase can't fix it either. After 6 months, I gave up.

UseN1bble, 2025-07-19:
The GET request is functioning correctly; however, the POST request is resulting in a 401 Unauthorized error.
I may attempt to contact Enphase or an installer again. They might be able to reset the installation.
Theoretically, I should be able to perform this action myself with the DIY account, but I am concerned about encountering difficulties. This issue is becoming more and more a matter of principle.

fobrs, 2025-07-19:
Yes, I understand. That's why I recently have got a metered envoy installed. Now I have full control over production and export.
Today I have set the export limit to 250 W and slew rate to 100 W/s. Also using a Homewizard Plugin Battery.
This works great: [https://solar.bigroses.net/7-19-2025](url)

deanillfeld, 2025-10-20:
I think Enphase might have broken something recently, even when i use Enlighten Manager to set a production limit nothing seems to happen, they dont even appear in Enlighten Manager. Export limits work fine. 

My envoy event log doesn't go back far enough but I'm fairly sure i had this working in the past by just setting `export_limit: false`

lawther, 2025-11-11:
Good to hear it's started working again as I have the same firmware. Can you tell me where you go to see the 'Production Limit Active' logs in your screenshot? Thanks!

deanillfeld, 2025-11-11:
You can see them on this page.

https://envoy/installer/setup/home#more/events

lawther, 2025-11-12:
Thanks! I hadn't seen these events before, I can't seem to find them in Enlighten. I also found another local endpoint https://envoy/event, that looks like the same info. 

My Envoy is being super weird with this API. I can see the event "Power export limiting active: Set Wed Nov 12, 2025 01:02 PM EST" - this was set remotely by my power company (Amber) to curtail production. However, power production was not affected. Then Amber went to turn production back on: "Power export limiting active: Clear Wed Nov 12, 2025 04:00 PM EST", and immediately the system stopped producing power (at 4:00:05 according to Home Assistant). I checked /ivp/ss/dpel at about 4:09PM, and it was correctly 
"{
    "date": "2025-11-12 04:59:59",
    "dynamic_pel_settings": {
        "enable": false
    },
    "filename": "site_settings",
    "version": "00.00.01"
}"

So I manually sent a POST using curl to /ivp/ss/dpel with the same JSON to disable power limiting, and production started back up again.

So my Envoy almost seems to do the previous thing it was told?

lawther, 2025-11-29:
Update - Enphase upgraded my firmware to D8.3.5232 (f18f5b), and this API is now working reliably. I did not ask for the upgrade, it just happened.

Prior to the upgrade, I had managed to get something sort of working, with the note from @jurrid in https://github.com/Matthew1471/Enphase-API/issues/31#issuecomment-3411834342 being key - the logic for 'export_limit' was inverted for me. The Envoy was also extremely sluggish to react to the PEL commands, often taking between 1.5 to 5 minutes before production changed. So I had to just write a loop, sending the command every 30 seconds and eventually the Envoy would react.

Now that I'm on D8.3.5232, the 'export_limit' logic has changed to be what I expect is correct (that is 'true' means apply an export limit, and 'false' means 'apply a production limit').  I am only using 'export_limit=false', but in my limited testing, 'export limit=true' appears to work with the right conditions (when my battery is disconnected or 100% full AND I set a non-zero export limit). The Envoy is now reacting within about 10-20 seconds which is so much better.

I still have no idea what the meaning of the date field is. Like others, I simply don't send it, and things seem to work fine.

Matthew1471, 2025-11-29:
Apologies for not being able to reply sooner. [All /ivp/ss/* calls get passed to a `site_set` binary](https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/Architecture.adoc#site-settings-handler) rather than regular Ruby code... if anyone is handy with arm dissassembly/decompilation I can share the binary (/opt/agf/bin/site_set) that maps to it.

I don't have Enphase battery (only solar) so my initial documentation of "Site" (aka. Grid) settings are a bit poor (I initially documented by probing - found I had access to the source code later). I have the Ruby source code for a lot of the endpoints but some of the others do just defer to a binary.

I'll run a search on "ss/dpel" and see if I can find any references to it in my cut of the firmware (and apps) too.

Matthew1471, 2025-11-30:
Toolkit looks like it has the answers:

classes3.dex\sources\com\enphase\installer\model\remote\EnvoyApiClientHelper.java:

```
        @GET("/ivp/ss/dpel")
        Call<PELSettings> getPELSettings();
...

        @POST("/ivp/ss/dpel")
        Call<ResponseBody> setPELSettings(@Body DynamicPELSettingsRequest dynamicPELSettingsRequest);
```

So it's looking like you're after what `DynamicPELSettingsRequest` and `PELSettings` serialises as.

I have an older version of Toolkit to hand 3.30.0 and that's giving:

```
    @SerializedName("enable")
    private final boolean enable;
    @SerializedName("enable_dynamic_limiting")
    private final Boolean enableDynamicLimiting;
    @SerializedName("export_limit")
    private final Boolean exportLimit;
    @SerializedName("installed_capacity")
    private final Integer installedDcCapacity;
    @SerializedName("limit_value_W")
    private final Integer limitValue;
    @SerializedName("relay_config")
    private final RelayConfig relayConfig;
    @LocalField
    @PrimaryKey
    private Long systemId;
```

and RelayConfig:

```
    @SerializedName("default_pct_limit")
    private final Integer defaultPctLimit;
    @SerializedName("limit_levels")
    private final List<LimitLevel> limitLevels;
```

and LimitLevel:

```
    @SerializedName("pct_limit")
    private final int pctLimit;
    @SerializedName("relays")
    private final String relays;
```

I'm not seeing any other params.. but this can't be correct as you reference "filename", "version" and "date". Perhaps this has undergone more changes since Tooolkit 3.30.0 so I'll grab a more recent copy.